csinko/io-master #27: Determine what pins we need for I/O

csinko, 2020-01-16:
Finalize what pins we are going to use

codeThatThinks, 2020-01-31:
The pin allocation spreadsheet lists all the pins we are going to use.

TLDR:
- ULPI interface used to communicate with USB PHY, these pins cannot be reassigned
- PORTD is used only for high speed digital outputs to the level shifter (low byte is IO_x_OUT, high byte is IO_x_TRIS)
- PORTE is used only for high speed digital inputs from the level shifter (low byte is NC, high byte is IO_x_IN)
- There is a dedicated hardware timer that is tied to each high speed digital output used for clock generation. Simply enable tristate for the PORTD pin and disable tristate for the timer output pin.
- I2C1 is used for communication to all level shifter DACs.
- USART1 will be broken out with test points in case it needs to be used for debugging/DFU mode
- DAC_OUT1 is used to control the adjustable regualtor for powering the target device

closed 2020-01-31 by codeThatThinks

codeThatThinks, 2020-02-13:
@WhoDaWhatNow I've changed some of the pin assignments in order to make the board layout easier. Please review the spreadsheet on drive.