csinko/io-master #29: USB PHY circuit design

codeThatThinks, 2020-01-16:
Choose a USB PHY from Digikey so that we can used USB 2.0 High Speed on the microcontroller. Draw up a schematic with USB connector, USB protection, PHY chip, and microcontroller interface.

csinko, 2020-01-26:
I would like to review which design is selected before we order it.  There's many options for the USB interface that I am using and I want to make sure what we both use are compatible with eachother.

I would expect it to be regular USB HS, but I am seeing a lot of stuff relating to OTG, which seems odd.  Its possible that portion of the library is used for other functions, the pins are shared, or something of that sort.

nuntch, 2020-01-28:
Schematic is done. Waiting for review on 1/30.

codeThatThinks, 2020-01-30:
ST refers to the built-in USB peripheral on the MCU as "USB OTG" even though it supports a variety of USB modes, including host, device, and OTG (switchable between host/device).

I/O Master will likely use the HS Device mode with an external PHY. The Microchip USB3300 PHY uses a ULPI interface to communicate with the MCU. This interface uses different pins and does not include the MCU's USB D+ and D-. We will include test points for the MCU D+ and D- pins in case we ever want to use the built-in USB PHY for full speed.

The pin allocation spreadsheet lists the pins used for the ULPI interface.

closed 2020-01-30 by codeThatThinks